ariel-os/ariel-os #140: Tracking formal verification status

emmanuelsearch, 2024-03-05:
Based on the current use of [hax](https://github.com/hacspec/hax) and with regards to our draft [manifesto](https://future-proof-iot.github.io/RIOT-rs/dev/), the question arises of where/how to track where we currently are at, now that initial integration has begun e.g. with [PR126](https://github.com/future-proof-iot/RIOT-rs/pull/126).

emmanuelsearch, 2024-03-05:
to give a concrete example: based on https://github.com/future-proof-iot/RIOT-rs/blob/main/src/riot-rs-runqueue/proofs/fstar/extraction/Makefile how/where should we express (for laypeople) what verification(s) are currently provided by CI for the scheduler?

emmanuelsearch, 2024-03-05:
@W95Psp @franziskuskiefer obviously we'd very much value your suggestions on this ;)

franziskuskiefer, 2024-03-11:
So do we have panic freedom already? If yes, then you should think about the contracts (pre- and post-conditions) the functions have and if there's something interesting that should be proven.
In parallel, adding more crates to the extraction, lax typechecking, etc. pipeline.

To the documentation, I think panic freedom should be documented when the extracted code typechecks in F*.
Additional properties should then come from the contracts the functions get and get described accordingly.

franziskuskiefer, 2024-03-11:
I think the easiest description is

- lax typechecking -> valid F* code is generated and all dependencies (like from the Rust core library) are defined
- typechecking -> panic freedom (F* proved that no panic occurs in the code)