ariel-os/ariel-os #156: threads: save callee-save registers on stack instead of `Threads::high_reg`

elenaf9, 2024-03-07:
Currently, the scheduler implementation for cortex-m saves the callee-save/ "high registers" in a separate field in the `Thread` struct. This field is an array with exactly 8 elements for the 8 register values (r4-r11).
This might not work well for other architectures (like RISC-V) since the number of registers that should be saved is different.

I would propose that we save the registers on the stack instead, i.e. pushing old register state before switching the sp, and then popping the new register state. RIOT(-c) is also using the stack for this [^1] [^2]

@kaspar030 would there be any issue if we do it this way and/ or was there a specific reason why you used a separate field?

[^1]: https://github.com/RIOT-OS/RIOT/blob/1bf643dd5aaa65ee0a653cf503ff5d65be76a20c/cpu/cortexm_common/thread_arch.c#L20-L83
[^2]: https://github.com/RIOT-OS/RIOT/blob/1bf643dd5aaa65ee0a653cf503ff5d65be76a20c/cpu/riscv_common/thread_arch.c#L41-L44

kaspar030, 2024-03-07:
> Currently, the scheduler implementation for cortex-m saves the callee-save/ "high registers" in a separate field in the `Thread` struct. This field is an array with exactly 8 elements for the 8 [...]

May I suggest to make that field architecture dependent?
It might make sense in any case to allow architectures to store data along with the threads.

``` Rust
struct Thread {
 ...
 arch: ThreadArch,
 }
 ```
 
 ... with an cortex-m ThreadArch that holds the high regs.
 
 

> @kaspar030 would there be any issue if we do it this way and/ or was there a specific reason why you used a separate field?

I hoped it would make some things easier. E.g, there's always a `high_regs` for each thread, even if those fields were never pushed to the thread. In theory, if a thread causes a voluntary context switch (e.g., by yielding or through self-"inflicted" blocking), it's possible to just not store those registers, and on next context restore, restore obsolete values.

In the end, I don't think it matters much were we store those registers. My impression is that the RIOT-c scheduling assembly is quite a bit longer than the RIOT-rs one, but that can have any number of reasons. :)

kaspar030, 2024-03-07:
> May I suggest to make that field architecture dependent?

If we do this, risc-v doesn't have to use the Thread struct to store its (high) registers.

elenaf9, 2024-03-07:
> May I suggest to make that field architecture dependent?
It might make sense in any case to allow architectures to store data along with the threads.

Okay :+1:. 
Included in #159.

kaspar030, 2024-03-10:
I'll close this for now, the conclusion being, it doesn't matter for ARM, and #159 brings abstraction so it won't matter for any architecture.

closed 2024-03-10 by kaspar030