ariel-os/ariel-os #199: nrf52840dk: USB issue in examples/embassy-net-usb

emmanuelsearch, 2024-03-19:
on my Ubuntu machine, when I run 

laze -C examples/embassy-net-udp/ build -b nrf52840dk -s probe-rs-run run

it compiles fine and flashes fine, and claims to be 'Listening on UDP:1234...'
but no interface shows up in ifconfig, and the only thing that pops up in lsusb is: 

Bus 001 Device 021: ID 1366:1051 SEGGER J-Link

In effect, I guess it is no surprise that pinging does not work as expected.
Can someone reproduce?

(Starting point: building, flashing and running examples/hello-world works fine on this board)

kaspar030, 2024-03-19:
Did you also connect the USB user port with a second cable?

emmanuelsearch, 2024-03-19:
I did. Now I get an USB Ethernet poping up indeed. ifconfig gives an IPv6 only interface only:

enx8a8888888888: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::bbdb:3ec8:e926:3545  prefixlen 64  scopeid 0x20<link>
        ether 8a:88:88:88:88:88  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19  bytes 3001 (3.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

emmanuelsearch, 2024-03-19:
but according to the [README](https://github.com/future-proof-iot/RIOT-rs/blob/main/examples/embassy-net-udp/README.md) we should ping 10.42.0.61

kaspar030, 2024-03-19:
> but according to the [README](https://github.com/future-proof-iot/RIOT-rs/blob/main/examples/embassy-net-udp/README.md) we should ping 10.42.0.61

That readme is only telling half of the truth. Indeed the node listens to that IP address, but your laptop is not in the same network.
You'll need to set up an address:

```shell
$ ip address add 10.42.0.1/24 dev <ACTUAL DEVICE NAME>
$ ip link set up dev <ACTUAL DEVICE NAME>
```

emmanuelsearch, 2024-03-19:
OK thanks a bunch. Got it to run indeed after setup with

$ (sudo) ip addr add 10.42.0.1/24 dev <DEVICE_NAME>
$ (sudo)  ip link set dev <DEVICE_NAME> up

then I got indeed:

$ ping -I <DEVICE_NAME> 10.42.0.61
PING 10.42.0.61 (10.42.0.61) from 10.42.0.1 enx8a8888888888: 56(84) bytes of data.
64 bytes from 10.42.0.61: icmp_seq=1 ttl=64 time=1.56 ms
64 bytes from 10.42.0.61: icmp_seq=2 ttl=64 time=0.942 ms
64 bytes from 10.42.0.61: icmp_seq=3 ttl=64 time=1.04 ms
64 bytes from 10.42.0.61: icmp_seq=4 ttl=64 time=1.02 ms
64 bytes from 10.42.0.61: icmp_seq=5 ttl=64 time=1.03 ms
64 bytes from 10.42.0.61: icmp_seq=6 ttl=64 time=1.06 ms

emmanuelsearch, 2024-03-19:
I guess this exchange could go verbatim into the example's [README](https://github.com/future-proof-iot/RIOT-rs/blob/main/examples/embassy-net-udp/README.md)

kaspar030, 2024-03-19:
> I guess this exchange could go verbatim into the example's [README](https://github.com/future-proof-iot/RIOT-rs/blob/main/examples/embassy-net-udp/README.md)

At least the info for the USB user port cable, and that the network interface needs configuring. Both are only valid for a (usb) ethernet connection, though. Wifi doesn't need the cable, and should rather disable "override-network-config" so it uses dhcp. And the example should print which IP address it is actually using. :smirk:

kaspar030, 2024-03-19:
And this should go a level higher, as this is the same in all three networking examples.

closed 2024-03-20 by emmanuelsearch