ariel-os/ariel-os #298: tracking: use of unstable features

kaspar030, 2024-05-21:
We're aiming to build with stable Rust.

Currently, we're using these unstable features:

1. can't do without:
  - [ ] [`used_with_arg`](https://github.com/rust-lang/rust/issues/93798): this is used by linkme for its distributed slices

2. could do without:
- [ ] [`type_alias_impl_trait`](https://github.com/rust-lang/rust/issues/63063) => embassy uses this or pre-allocate futures memory
- [ ] `const_mut_refs`: used by ringbuffer
- [ ] `custom_test_frameworks`: will go when we have migrated to `defmt-test`
- [ ] `doc_auto_cfg`
- [ ] `error_in_core`
- [ ] `fn_traits`
- [ ] `inline_const`
- [ ] [naked_functions](https://github.com/rust-lang/rust/issues/90957): used by `riot-rs-threads`

josephcsible, 2024-12-23:
>     * [ ]  [`negative_impls`](https://github.com/rust-lang/rust/issues/68318); used for `!Send` on `MutexGuard`

Could you get rid of that by adding something like `_not_send: PhantomData<*const ()>` to `MutexGuard` instead?

kaspar030, 2025-03-18:
Where are `fn_traits` used?

```shell
❯ git grep fn_traits
ariel-os on  main took 24ms 
❯ 
```

ROMemories, 2025-03-18:
> Where are `fn_traits` used?

I'm not sure, I searched for it in my fork as well (in case that was for the sensor abstraction), but couldn't find anything. I removed it from the list.

kaspar030, 2025-03-29:
> This leaves only `used_with_arg`, and that we might manage by virtue of having good control of linker scripts ([#639](https://github.com/ariel-os/ariel-os/issues/639)?)

I'll give solving `used_with_arg` another shot.

kaspar030, 2025-03-29:
`ariel-os-debug-log` is using `macro_metavar_expr`