ariel-os/ariel-os #351: `examples/embassy` doesn't work

elenaf9, 2024-07-08:
### Summary

I just tried running the `embassy` example and it didn't work as expected. Specifically, `await`ing the timer never returned. 
The example seems quite old, is it outdated? Or any other idea why it stopped working?

### Relevant log output

```shell
riot_rs_rt::startup()
rpi-pico::init()
riot-rs-embassy::init(): using interrupt mode executor
riot-rs-embassy::init_task()
riot-rs-embassy::init_task() done
Hello from main()! Running on a rpi-pico board.
async_task() signalling
before timer()
now=0ms threadtest() val=0
```

### Hardware/ Setup

- Board: `rpi-pico` - I don't have a nrf board with me right now so I can't check if it works there
- Branch: current main (29b7310)

### Possible Solution

This might be an issue around timers in `embassy_rp`; I only tested this example in the first place because I ran into some timer-related issues when running a benchmark for my WIP multicore implementation.

### Would you like to work on fixing this bug?

Maybe

kaspar030, 2024-07-08:
> However, I _can_ reproduce your output on an Raspberry Pi Pico W

Same!

elenaf9, 2024-07-08:
Hmm, but generally the timers work for the RP2040, at least with the `executor-thread`. The below code works:

```rs
#![no_main]
#![no_std]
#![feature(type_alias_impl_trait)]
#![feature(used_with_arg)]

use embassy_time::{Duration, Timer, TICK_HZ};
use riot_rs::{
    debug::println, embassy::thread_executor::Executor, static_cell::make_static,
    thread::ThreadId,
};

// Task started in thread0
#[riot_rs::task(pool_size=1)]
async fn task0() {
    println!("task0");
    Timer::after(Duration::from_ticks(TICK_HZ / 10)).await;
    println!("task0 done");
}

#[riot_rs::thread(autostart)]
fn thread0() {
    println!("thread0 {:?}", riot_rs::thread::current_pid().unwrap());
    let executor = make_static!(Executor::new());
    executor.run(|spawner| {
        spawner.must_spawn(task0());
    });
}

// Autostarted task.
#[riot_rs::task(autostart)]
async fn task1() {
    println!("task1");
    Timer::after(Duration::from_secs(1)).await;
    println!("task1 done");
}

```

Maybe the issue is related to the `InterruptExecutor` that is used in the above example?