ariel-os/ariel-os #372: nRF53 HWRNG

chrysn, 2024-07-31:
### Description

Compared to nRF52, nRF53 has no dedicated HWRNG any more, but provides one through the cryptocell, for which documentation was [recently published](https://docs.nordicsemi.com/bundle/ps_nrf5340/page/cryptocell.html#ariaid-title18) (before, it was only available through a nonfree vendor SDK).

The documentation should in theory suffice to implement a HWRNG.

### Motivation

Having an RNG seed is required to run EDHOC/CoAP, which we currently can not on nRF53.

### Requirements

Implement the laze module hwrng for the context nrf53, currently in `src/riot-rs-embassy/src/arch/nrf/hwrng.rs`

### Are you planning to do it yourself in a pull request?

Maybe

chrysn, 2024-07-31:
Practically, this is blocked by https://github.com/NordicSemiconductor/nrfx/issues/123 -- as long as there are no new SVDs, we won't get an update in https://github.com/nrf-rs/nrf-pacs, and I don't plan on doing this on foot.

nponsard, 2026-02-16:
This is implemented in #1804, it will need some time before getting merged as `embassy-nrf` needs to be updated (both a new version needs to be cut upstream and the new version needs to be used in Ariel).

closed 2026-10-02 by nponsard