ariel-os/ariel-os #410: cleanup examples

kaspar030, 2024-09-11:
### Description

For each of them,

- minimize
- find proper name
- polish
- ...

### Motivation

...

### Current Implementation

...

### Are you planning to do it yourself in a pull request?

Maybe

elenaf9, 2024-09-13:
- Be consistent in the usage of laze modules vs `riot_rs` features.

emmanuelsearch, 2024-09-16:
Consider adding examples on
- basics: eg timers, sensor abstraction etc. inspired by RIOT-course https://riot-os.github.io/riot-course/slides/03-riot-basics/#11
- security: hash, sign, encrypt etc. inspired by RIOT-course on sec. https://riot-os.github.io/riot-course/slides/06-security-with-riot/#14

bergzand, 2025-02-11:
sufficiently fixed with #792 merged

closed 2025-02-11 by bergzand