ariel-os/ariel-os #418: `examples/coap` (liboscore) on armv6m CI build failure on CI

kaspar030, 2024-09-16:
### Summary

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/runner/.rustup/toolchains/nightly-2024-05-20-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/runner/.cargo/bin:/opt/hostedtoolcache/sccache/0.8.1/x64:/snap/bin:/home/runner/.local/bin:/opt/pipx_bin:/home/runner/.cargo/bin:/home/runner/.config/composer/vendor/bin:/usr/local/.ghcup/bin:/home/runner/.dotnet/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcTxUJH8/symbols.o" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/deps/coap-1a5ba26ce4603683.coap.980900a94a691917-cgu.0.rcgu.o" "--as-needed" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/deps" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/release/deps" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/defmt-80592f337d3d523b/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/liboscore-1869f3cbdcc9cefc/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/cortex-m-15f9f17cccca1b61/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/riot-rs-rt-40131ddd18c2793d/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/cortex-m-rt-cf61485e44677a0d/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/rp2040-7fe733d862f4f47d/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/embassy-rp-30e866bf5011cef2/out" "-L" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/build/rp-pac-57c39bb19e30139e/out" "-L" "/home/runner/.rustup/toolchains/nightly-2024-05-20-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-Bstatic" "/tmp/rustcTxUJH8/libliboscore-c1002fd4db71cefd.rlib" "/tmp/rustcTxUJH8/libcortex_m-9c72b8b9f3b7a158.rlib" "/home/runner/.rustup/toolchains/nightly-2024-05-20-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcompiler_builtins-9573e01946a4d685.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/home/runner/.rustup/toolchains/nightly-2024-05-20-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-o" "/home/runner/work/RIOT-rs/RIOT-rs/examples/coap/../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/deps/coap-1a5ba26ce4603683" "--gc-sections" "--nmagic" "--no-eh-frame-hdr" "-Tlinkme.x" "-Tlink.x" "-Tdevice.x" "-Tisr_stack.x" "-Tlink-rp.x" "-Tdefmt.x"
  = note: rust-lld: error: undefined symbol: __gnu_thumb1_case_uqi
          >>> referenced by oscore_message.c:52 (c-src/oscore_message.c:52)
          >>>               905f1f7c3b7e961f-oscore_message.o:(get_option_behaviour) in archive /tmp/rustcTxUJH8/libliboscore-c1002fd4db71cefd.rlib
          

warning: `coap` (bin "coap") generated 6 warnings
error: could not compile `coap` (bin "coap") due to 1 previous error; 6 warnings emitted
FAILED: build/bin/rpi-pico/coap/coap.elf 
cd examples/coap && OPENOCD_ARGS="-f interface/cmsis-dap.cfg -f target/rp2040.cfg" SCRIPTS=../../scripts CONFIG_BOARD=rpi-pico CARGO_BUILD_TARGET=thumbv6m-none-eabi CARGO_TARGET_THUMBV6M_NONE_EABI_RUNNER='probe-rs run --protocol=swd --chip RP2040' CARGO_TARGET_THUMBV6M_NONE_EABI_RUSTFLAGS="-Clink-arg=--nmagic -Clink-arg=--no-eh-frame-hdr --cfg builder=\"rpi-pico\" --cfg context=\"rpi-pico\" --cfg context=\"riot-rs\" --cfg context=\"rp\" --cfg context=\"rp2040\" -Clink-arg=-Tlinkme.x -Clink-arg=-Tlink.x -Clink-arg=-Tdevice.x -Clink-arg=-Tisr_stack.x --cfg context=\"cortex-m\" --cfg armv6m -Clink-arg=-Tlink-rp.x -Clink-arg=-Tdefmt.x --cfg capability=\"hw/usb-device-port\" -Cembed-bitcode=yes -Clto=fat -Ccodegen-units=1" CARGO_TARGET_DIR=../../build/bin/rpi-pico/coap/cargo RUSTC_WRAPPER="" DEFMT_LOG=info, CONFIG_ISR_STACKSIZE=16384 cargo  build --release --features=riot-rs-boards/rpi-pico,riot-rs/debug-console,riot-rs/defmt,riot-rs/executor-interrupt,riot-rs/hwrng,riot-rs/random,riot-rs/usb-ethernet && cp ../../build/bin/rpi-pico/coap/cargo/thumbv6m-none-eabi/release/coap ../../build/bin/rpi-pico/coap/coap.elf
laze: error: ninja exited with code 1
ninja: build stopped: subcommand failed.
Error: Process completed with exit code 1.

### Would you like to work on fixing this bug?

Maybe

kaspar030, 2024-09-16:
This was the failing CI run: https://github.com/future-proof-iot/RIOT-rs/actions/runs/10881859932/job/30192896887?pr=413

chrysn, 2024-09-16:
A quick (or quack?) duck duck go run did not come up with full solutions. Potential workarounds are setting `-fno-builtin-function` or another optimization level, but those do not seem to address the root cause. Another avenue to check is whether we're building with GCC rather than clang for some reason -- it'd make sense to use clang given we're later also linking with it (maybe versions mismatch?).

chrysn, 2025-01-13:
Can you look into whether consistency of LLVMs fixes this? I sometimes use `LLVM_CONFIG_PATH=/usr/bin/llvm-config-19 LIBCLANG_PATH=/usr/lib/llvm-19/lib/` on multi-LLVM systems.

kaspar030, 2025-03-26:
This was fixed by #826.

closed 2025-03-26 by kaspar030