ariel-os/ariel-os #610: Create an FAQ (gathering intial entries)

chrysn, 2024-12-10:
### Description

We don't have an FAQ right now.

Where should that go? Book?

### Motivation

"SBOM" came up every other weekly, let's close that by adding an FAQ item.

### Current Implementation

Let's collect here:

* Q: Do you have an SBOM?

  A: All code that goes into a build is gathered, compiled and linked through Rust's built in cargo package manager, stemming from its default package repository [crates.io](https://crates.io/) or, during development, upstream git repositories. The crate descriptions provide metadata about versions, licenses, upstream web pages and much more. Some of that metadata is checked at CI time; for example, we use [`cargo deny`](https://docs.rs/cargo-deny/latest/cargo_deny/) to vet our licenses. Tools are available to extract information from there (like for any other Rust project), for example [`cargo sbom`](https://crates.io/crates/cargo-sbom).

  Note that not all dependencies of Ariel OS are always built for all applications. If you want to evaluate properties of a particular build, tooling is being developed that simplifies calling custom Cargo commands on a particular narrow configuration.

### Are you planning to do it yourself in a pull request?

Maybe

chrysn, 2024-12-10:
One more potential entry:

* Q: What is your relation to RIOT OS?

  A: Ariel OS was started by RIOT maintainers as a rewrite of specific components of RIOT OS, with the goal of applying memory safety and modern language concepts to RIOT. The project shifted towards not providing the identical C APIs, but still builds on concepts and experience from RIOT -- in some sense it is a rewrite of RIOT, but trying to translate meaning rather than words.

  There is active collaboration between the operating systems. While applications can currently not be ported between Ariel OS and RIOT OS, there is shared code used in examples of both, and we hope to enable a subset of the possible applications to be portable between both platforms. In addition to shared code, we also share infrastructure, have common goals, and meet as part of the RIOT community.

kaspar030, 2025-02-11:
## How does Ariel OS compare to ...

### Embassy

We love Embassy the way it is. Ariel OS wouldn't be possible without it. We'll keep building on top of Embassy, we'll track Embassy's develpment, and we'll work on upstreaming all changes that upstream deems worthy.

Ariel OS has Embassy at its heart. Embassy is used for hardware abstraction, async executor, networking, timers, ...

On top of that, Ariel OS adds

- a multi-core enabled preemptive scheduler
  - this allows mix-and-match of sync and async code within an application
- abstracted peripheral APIs, increasing portability
  - while Embassy's HALs implement the embedded-hal(-async) traits, the initialization API is slightly different for each MCU family (it fits perfectly for the corresponsing hardware). Ariel OS abstracts that provides an API that is identical *across* MCU families.
- pre-integrated modules like networking, storage that are ready to use
  - where Embassy and the eco system provide all the building blocks (embassy-net, rand), setup and initialization needs per-MCU copy&pasting and combining from the examples and resources. Ariel tries to improve on that, e.g., setting up the hwrng or network stack centrally, making it ready to use for applications.
- a modularized system initialization scheme
  - where Embassy leaves full control to the application developer by requiring all set up to be done in the application's `main()`, by the application developer, Ariel does a lot of initialization (e.g., network stack, rng, storage) internally, and hands a ready-to-use system to the developer
- a meta build system that handles supporting many different target devices
  - Ariel OS wraps Cargo in [laze](https://laze-build.org) in order to manage all the build-system configuration that Cargo cannot handle itself, like `--target`, the used runner, probe-rs configuration, linker setting, that are usually hard-coded per board. laze simplifies building for multiple target boards.

Practically, Ariel OS moves some of the control of the classical main loop most Embassy applications have, and handles a lot of the "generic system" stuff itself. Actual applications contain less system bringup, less boilerplate, and can concentrate on business logic.

### Tock

Tock has been around for more than six years, and was probably the first full-blown Rust embedded OS. It is designed around the idea of isolating the OS from applications, allowing to run untrusted applications.

Ariel OS has a traditional "library OS" design, where the application(s) and the OS are built together, sharing the same memory space.

Tock OS was designed before Rust async support existed, and thus does not make use of it. Ariel OS does, by having Embassy at its heart, fully embrace the Rust async way for its system components.

Tock OS developed most code in-house and has little third-party dependencies, because it basically grew up before there was a Rust Embedded ecosystem worth mentioning. Ariel OS on the contrary makes full use of the by now extensive ecosystem, integrating and curating the often high-quality crates, and ties them together.

emmanuelsearch, 2025-02-11:
Another candidate for FAQ entry: how does Ariel compare to other OS out there? Basically #345

emmanuelsearch, 2025-02-13:
solved with #794

PR #794 (2025-02-13): docs(FAQ): add FAQ.md
# Description

This PR adds an FAQ.

## Issues/PRs references

Closes #610.
Closes #345.

- [x] I have cleaned up my commit history and squashed fixup commits.
- [x] I have followed the [Coding Conventions](https://ariel-os.github.io/ariel-os/dev/docs/book/coding-conventions.html).
- [x] I have performed a self-review of my own code.
- [x] I have made corresponding changes to the documentation.
closed 2025-02-13 by emmanuelsearch