ariel-os/ariel-os #667: build: lingering build issues around liboscore

kaspar030, 2025-01-13:
This came up at the today's weekly:

- Q: kaspar reports some lingering building issues with coapcore (doing full builds, maybe LLVM version issues...)
  - A: Try this environment: `LLVM_CONFIG_PATH=/usr/bin/llvm-config-19 LIBCLANG_PATH=/usr/lib/llvm-19/lib/`
  - Maybe this is something only laze / the build environment can decide.
  - → Kaspar will create issue

chrysn, 2025-01-14:
Is this https://github.com/ariel-os/ariel-os/issues/418 or more?

kaspar030, 2025-01-14:
I think it is more:

```
[45/444] CARGO build/bin/nrf52840dk/coap-client/coap-client.elf                                                                          
   Compiling liboscore v0.2.2                                                                                                            
error: failed to run custom build command for `liboscore v0.2.2`                                                                         
                                                                                                                                         
Caused by:                                                                                                                                                                                                                                                                        
  process didn't exit successfully: `/home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/release/build/liboscore-8dbd681f8bef6e47/build-script-build` (exit status: 101)                                                                     
  --- stdout                                                                                                                                                                                                                                                                      
  cargo:PLATFORMHEADERS=/home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers                                                                              
  cargo:rerun-if-env-changed=TARGET                                                                                                                                                                                                                                               
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_thumbv7em-none-eabi                                                                                                                                                                                                         
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_thumbv7em_none_eabi                                                                                                                                                                                                         
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS                                                                                                                                                                                                                             
  cargo:rerun-if-changed=oscore_all_headers.h                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                                  
  --- stderr                                                                                                                                                                                                                                                                      
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/crypto_type.h:10:9: error: unknown type name 'uint32_t'                              
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/crypto_type.h:13:5: error: unknown type name 'uint64_t'                              
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/crypto_type.h:17:5: error: unknown type name 'uint64_t'                              
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/crypto_type.h:37:9: error: unknown type name 'uint8_t'
  c-src/include/oscore/helpers.h:72:5: error: unknown type name 'uint8_t'                                                                
  c-src/include/oscore/helpers.h:74:5: error: unknown type name 'uint8_t'                                                                
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/msg_type.h:26:9: error: unknown type name 'uint8_t'
  /home/kaspar/src/own/rust/ariel-os/examples/coap-client/../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/build/liboscore-207cb1a01f045b8d/out/rust-built-headers/oscore_native/msg_type.h:50:3: error: unknown type name 'uint64_t'
[... cut ...]
debug-console,ariel-os/hwrng,ariel-os/random,ariel-os/usb,ariel-os/usb-ethernet && cp ../../build/bin/nrf52840dk/cargo/thumbv7em-none-eabi/release/coap-client ../../build/bin/nrf52840dk/coap-client/coap-client.elf
ninja: build stopped: subcommand failed.                                                                                                 
laze: error: ninja exited with code 1                               
```

> Try this environment: LLVM_CONFIG_PATH=/usr/bin/llvm-config-19 LIBCLANG_PATH=/usr/lib/llvm-19/lib/

I'm on llvm 19 since today (Arch just updated), I got the same errors. Those paths don't exist, but I only have the one llvm installed. (maybe another with the esp stuff?)

kaspar030, 2025-01-22:
I also get this when doing `laze build -b host -mg test`:

```
laze: executing task test for builder host bin crates/coapcore                 
warning: Patch `esp-storage v0.4.0 (https://github.com/ariel-os/esp-hal?branch=v0.23.1%2Bariel-os#6569d3a6)` was not used in the crate graph.                  
Check that the patched package version and available features are compatible                                                                                   
with the dependency requirements. If the patch has a different version from                                                                                    
what is locked in the Cargo.lock file, run `cargo update` to use the new                                                                                       
version. This may also occur with an optional dependency that is not enabled.                                                                                  
   Compiling liboscore v0.2.2                                                                                                                                  
error: failed to run custom build command for `liboscore v0.2.2`                                                                                               
                                                                                                                                                               
Caused by:                                                                                                                                                     
  process didn't exit successfully: `/home/kaspar/src/own/rust/ariel-os/target/debug/build/liboscore-7edcb388c81cd11f/build-script-build` (exit status: 101)   
  --- stdout                                                                                                                                                   
  cargo:PLATFORMHEADERS=/home/kaspar/src/own/rust/ariel-os/target/debug/build/liboscore-53e91de42affd688/out/rust-built-headers                                
  cargo:rerun-if-env-changed=TARGET                                                                                                                            
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_x86_64-unknown-linux-gnu                                                                                 
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS_x86_64_unknown_linux_gnu                                                                                 
  cargo:rerun-if-env-changed=BINDGEN_EXTRA_CLANG_ARGS                                                                                                          
  cargo:rerun-if-changed=oscore_all_headers.h                                                                                                                  
                                                                                                                                                               
  --- stderr                                                                   
  thread 'main' panicked at /home/kaspar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.70.1/lib.rs:892:13:                                    
  assertion `left == right` failed: "x86_64-unknown-linux-gnu" "x86_64-unknown-linux-gnu"                                                                      
    left: 4                                                                                                                                                    
   right: 8                                                                                                                                                    
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace                                                                                
laze: the following tasks failed:
laze: task "test" on app "crates/coapcore" for builder "host"

chrysn, 2025-01-30:
for https://github.com/ariel-os/ariel-os/issues/667:

Starting to reproduce this here so I can make progress on it. Steps:

```
$ docker run --rm -it docker.io/archlinux
# pacman -Sy
# pacman -S git rustup clang ninja
# rustup default stable
# cargo install laze
# PATH=$PATH:/root/.cargo/bin/
# cd /tmp
# git clone https://github.com/ariel-os/ariel-os
# cd ariel-os
# laze build -b nrf52840dk -C examples/blinky
# laze build -b nrf52840dk -C examples/coap-server
```

Clang was installed because it is apparently in laze's dependencies; starting investigation from here.

"error occurred in cc-rs: Failed to find tool. Is `arm-none-eabi-gcc` installed?" -- but must it be GCC? After all, [the cc crate supports the common environment variables](https://docs.rs/cc/latest/cc/#external-configuration-via-environment-variables), so let's try:

```
# CC=clang laze build -b nrf52840dk -C examples/coap-server
```

That already builds -- problem solved? (Playing around a bit more with GCC, maybe there's also something to fix there.)

chrysn, 2025-01-30:
My main question (after digging a bit further but finding nothing easy on gcc) is: Is "If your GCC acts up, just set CC=clang" a viable solution here?

(Note that this particular issue is about the particular build issue @kaspar030 is experiencing)

kaspar030, 2025-02-14:
I just realized that I get this error when building for cortex-m, but having the LLVM_LIB path mungled by the espup provided `export-esp.sh`.

kaspar030, 2025-03-12:
https://github.com/ariel-os/ariel-os/pull/825 fixed this for me.

closed 2025-03-12 by kaspar030