ariel-os/ariel-os #785: Update documentation with STM32WBA55 reference

MaximeB5, 2025-02-10:
### Description

This enhancement proposal is not to forget to add to the documentation references to STM32WBA55 which was added following these 2 PRs https://github.com/ariel-os/ariel-os/pull/782 and https://github.com/ariel-os/ariel-os/pull/783 .

Package name in Ariel OS is st-nucleo-wba55.
The board is STM32WBA55CGUx.

As I'm not 100% sure about the fields listed here https://ariel-os.github.io/ariel-os/dev/docs/book/hardware_functionality_support.html , I prefer to open this ticket to make sure things are done the right way.

### Motivation

It'll add another STM32 to the existing list of supported devices, alongside STM32F401RETX, STM32H755ZITX and STM32WB55RGVX.

### Current Implementation

N/A.

### Are you planning to do it yourself in a pull request?

Yes

MaximeB5, 2025-02-10:
Do you have any recommendations regarding elements/code/etc not to be forgotten in the PR please ?

kaspar030, 2025-02-12:
We're also working on automating the whole process. The current idea is to mark features as supported if the build system builds them for a target, fully automatic. Needs some laze support, though.

PR #1064 (2025-05-28): docs(stm32): add STM32WBA55CG to the support matrix
# Description

Support for the Nucleo STM32WBA55 board was added in #782 and #783. This PR adds this board to the support matrix.

## Issues/PRs references

Closes #785.
Depends on #1057.

## Open Questions

## Change checklist

- [x] I have cleaned up my commit history and squashed fixup commits.
- [x] I have followed the [Coding Conventions](https://ariel-os.github.io/ariel-os/dev/docs/book/coding-conventions.html).
- [x] I have performed a self-review of my own code.
- [x] I have made corresponding changes to the documentation.