ariel-os/ariel-os #815: Use better marker key/value for storage

bergzand, 2025-02-13:
I think the [`MARKER_KEY`](https://github.com/ariel-os/ariel-os/blob/a6f025a3429b460b228af02e21b31390feed8bb0/src/ariel-os-storage/src/lib.rs#L80) can be improved upon. Either change to:

- A fully random key value, e.g. a `pwgen -s 10`
- A clear identifier that makes the origin and purpose clear, e.g. `ariel_init_mark`

From what I can see we only have to check *if* this key is present, can we change the value to the unit type (Seems to work with serde)?

kaspar030, 2025-02-13:
Maybe, a version? One that captures all the implementation details?

bergzand, 2025-02-13:
> Maybe, a version? One that captures all the implementation details?

Like the sequential-storage major version number? Auto-erase the storage when that gets incremented and backwards compatibility is gone? Maybe a bit on the dangerous side.

For the key or for the value btw?

kaspar030, 2025-02-13:
> Like the sequential-storage major version number? Auto-erase the storage when that gets incremented and backwards compatibility is gone? Maybe a bit on the dangerous side.

Yeah, probably right.

> For the key or for the value btw?

I was thinking value.

I unit type works, we can use that. The whole marker is there b/c without it, on corrupted (or uninitialized) flash, the first read/write just errors.

bergzand, 2025-02-13:
Any opinion on whether to use a randomized key or a 'clearly put there by Ariel OS' key?

chrysn, 2025-02-14:
Do we have the postcard hashes per value already? That'd significantly change what it means for any state to be compatible.

kaspar030, 2025-02-14:
> Do we have the postcard hashes per value already?

no. will bump on todo list.

chrysn, 2025-02-14:
> Values leftover from a previous firmware, as I think the entire flash is not always erased when flashing new firmware.

My impression was that having storage persisted across firmware versions was a desired property. I think we should make up our minds first on whether or not is the case, and then decide the implementation. (If it is not desired, an easy implementation might be to just include one more byte, say 0xff, in the firmware, or mark that area as to be erased and not written).

Then we can start with versioning topics.