ariel-os/ariel-os #880: nrf5340dk: flashing fails due to locked core

emmanuelsearch, 2025-03-02:
### Summary

Probably related with https://github.com/probe-rs/probe-rs/issues/3053

### Relevant log output

```shell
WARN probe_rs::vendor::nordicsemi::sequences::nrf: Core 1 is locked. Erase procedure will be started to unlock it.
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An operation could not be performed because it lacked the permission to do so: erase_all
```

### Would you like to work on fixing this bug?

No

chrysn, 2025-03-09:
Have you tried turning it off and on again? (Seriously: Unplugging helps with many cases of debuggers that get stuck.)

There is also a tool called "nrf-recover" (`cargo install nrf-recover`) that gets nRF devices out of some stuck situations.

Feedback on which of those two helped will help us understand why this happens.

kaspar030, 2025-03-09:
Does adding '-- --alloe-erase-all' help?

emmanuelsearch, 2025-03-19:
@chrysn unplug-replug does not help...

emmanuelsearch, 2025-03-19:
@kaspar030 adding '-- --allow-erase-all' does work!

emmanuelsearch, 2025-03-19:
You still get a couple of warnings though:

 ```
     Running `probe-rs run --protocol=swd --chip nrf5340_xxAA --preverify ../../build/bin/nrf5340dk/cargo/thumbv8m.main-none-eabi/release/hello-world --allow-erase-all`
 WARN probe_rs::vendor::nordicsemi::sequences::nrf: Core 1 is locked. Erase procedure will be started to unlock it.
 WARN probe_rs::architecture::arm::core::armv8m: The core is in locked up status as a result of an unrecoverable exception
 WARN probe_rs::architecture::arm::core::armv8m: The core is in locked up status as a result of an unrecoverable exception
      Erasing ✔ 100% [####################]  16.00 KiB @  28.76 KiB/s (took 1s)
  Programming ✔ 100% [####################]  16.00 KiB @  17.29 KiB/s (took 1s)      Finished in 1.91s
INFO  Hello World!
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    A timeout occurred during an operation.
```

nponsard, 2025-10-22:
> Are there any updates around this issue?

The nrf5340 is causing some issues with some versions of probe-rs, have you tried with probe-rs version `v0.27.0` ? 

Another thing to try is to use the [`nrfutil` tool](https://www.nordicsemi.com/Products/Development-tools/nRF-Util), using `nrfutil device recover` and `nrfutil device erase` before flashing. Had some success with it.

zhaolanhuang, 2025-10-22:
> The nrf5340 is causing some issues with some versions of probe-rs, have you tried with probe-rs version `v0.27.0` ?
>
> Another thing to try is to use the [`nrfutil` [...]

Downgrade probe-rs works fine (`v0.29.1` => `v0.27.0`). Thanks!