ariel-os/ariel-os #903: USB support on ESP32-S3

chrysn, 2025-03-12:
### Description

The hardware to do USB is present on those devices, and I think that's the largest contiguous area / largest single fix of red "X"es in our hardware matrix.

Also, this was [one of the first things that someone popping up on on our channel asked for](https://matrix.to/#/!cINbgbHLemsitmThSh:matrix.org/$h-MwRoxysdGRC9LrFS5cjpumHTOaMgdixXJsp7pcdNE?via=matrix.org&via=utwente.io&via=club1.fr).

### Motivation

Because it is there.

### Requirements

USB Ethernet would be a good starting demo.

### Are you planning to do it yourself in a pull request?

No

kaspar030, 2025-03-13:
I changed the subject to "USB ethernet support", as usb is actually  wired up and usb-serial works fine. There's an issue with usb-ethernet only, similar to #357.

chrysn, 2025-03-13:
Is the "X" at "user USB support" in [the table](https://ariel-os.github.io/ariel-os/dev/docs/book/hardware_functionality_support.html) wrong, then, or are those disabled because only some descriptors work (and thus the general case is not exposed)?

kaspar030, 2025-03-13:
No, I just checked, usb-serial also doesn't enumerate :(

kaspar030, 2025-10-16:
Confirmed and that matches documentation.

closed 2025-10-16 by kaspar030