ariel-os/ariel-os #904: laze messages for "you can't select that" are not helpful

chrysn, 2025-03-12:
### Summary

I just tried

```
$ laze build -b espressif-esp32-s3-devkitc-1 -C examples/udp-echo/  -s usb-ethernet
$ laze build -b espressif-esp32-s3-devkitc-1 -C examples/coap-server/  -s usb-ethernet
```

even though usb-ethernet is not supported on s3 (only checked after; see #903), and if I were a newcomer (well, even having spent some time on the project), it's not easy to figure out what is wrong.

### Relevant log output

```shell
configuring udp-echo for espressif-esp32-s3-devkitc-1
laze: not building binary "udp-echo" for builder "espressif-esp32-s3-devkitc-1"

Caused by:
    0: "udp-echo" cannot resolve "context::espressif-esp32-s3-devkitc-1"
    1: "context::espressif-esp32-s3-devkitc-1" cannot resolve "context::esp32-s3-wroom-1"
    2: "context::esp32-s3-wroom-1" cannot resolve "context::esp32s3"
    3: "context::esp32s3" cannot resolve "xtensa"
    4: "xtensa" is disabled/conflicted

configuring example-coap-server for espressif-esp32-s3-devkitc-1
laze: not building binary "example-coap-server" for builder "espressif-esp32-s3-devkitc-1"

Caused by:
    0: "example-coap-server" cannot resolve "context::espressif-esp32-s3-devkitc-1"
    1: "context::espressif-esp32-s3-devkitc-1" cannot resolve "context::esp32-s3-wroom-1"
    2: "context::esp32-s3-wroom-1" cannot resolve "context::esp32s3"
    3: "context::esp32s3" conflicts already provided "c-function-abort" (by "liboscore-provide-abort")
```

### Possible Solution

For the CoAP example, c-function-abort is provided by liboscore-provide-abort -- but there's no reason for laze to pick that, 3 is unjustified (but I guess liboscore-provide-abort was just not kicked out because esp32s3 was not selected successfully). In both cases, I'm missing the actually relevant information: that usb-ethernet is part of any conflict.

### Would you like to work on fixing this bug?

Maybe

chrysn, 2025-03-12:
I've filed this here and not in laze because somehow I think that this might be more related to how we use laze than to what laze does; if not, happy to keep tracking this in laze.

kaspar030, 2025-03-13:
> I've filed this here and not in laze because somehow I think that this might be more related to how we use laze than to what laze does; if not, happy to keep tracking this in laze.

This is a shortcoming in laze, I opened an issue [there](https://github.com/kaspar030/laze/issues/675). Laze doesn't track *why* a module was selected, so doesn't have the info here in combination with contexts disabling modules.

kaspar030, 2025-03-29:
@chrysn I think with laze `0.1.35` this is much better:

```
laze: not building binary "udp-echo" for builder "espressif-esp32-s3-devkitc-1"                                                                                
                                                                               
Caused by:                                                                                                                                                     
    0: "udp-echo" cannot resolve "usb-ethernet"                             
    1: "usb-ethernet" cannot resolve "usb"                                 
    2: "usb" cannot resolve "hw/usb-device-port"                        
    3: "hw/usb-device-port" cannot resolve "has_usb_device_port"               
    4: Could not resolve the following providers of "has_usb_device_port":
       1:                                                                                                                                                      
       "context::espressif-esp32-s3-devkitc-1" cannot resolve "context::esp32-s3-wroom-1"                                                                      
       "context::esp32-s3-wroom-1" cannot resolve "context::esp32s3"                                                                                           
       "context::esp32s3" cannot resolve "xtensa"                                                                                                              
       "xtensa" is disabled/conflicted by usb-ethernet                                                                                                         
       2:                                                                      
       "has_usb_device_port" cannot resolve "doc-only"                                                                                                         
       module "doc-only" not found
```

1. It now mentions that `usb-ethernet` cannot be resolved
2. It gives a hint that the reason is "xtensa".

I'll close this, feel free to reopen-

closed 2025-03-29 by kaspar030