ariel-os/ariel-os #934: threading-multicore: 2nd core fails on ESP32-S3

emmanuelsearch, 2025-03-23:
### Summary

The threading-multicore example prints out only the id of the first core, then hangs.

### Relevant log output

```shell

```

### Would you like to work on fixing this bug?

No

kaspar030, 2025-03-24:
Seems like this is a core affinity issue.

1. I get the same on rpi-pico with the thread executor
2. rpi-pico with interrupt executor works fine
3. disabling core affinity make it work on s3

kaspar030, 2025-03-24:
Turns out this is actually a case of "works as designed".

1. Both threads start, but instantly lock-block on the `THREAD_START_EVENT`.
2. Executor thread starts on core 0, initializes system, sets `THREAD_START_EVENT`.
3. Both threads get unlocked at the same time, with core 0 in use by executor thread.
4. Unclear why exactly (looks like a race) but thread 1 gets scheduled on core 1.
5. Thread 0 (which has its core affinity set to core 1) cannot be scheduled on core 1 because a thread with the same priority is already running there.

I'd love to get more clarity on the racyness in 4., as this changes when adding debug output.

Whether the scheduler should migrate a thread between cores if that would unlock another pinned thread, we need to discuss. I tend to "yes" actually, as the current behavior is surprising.

PR #937 (2025-03-25): fix(examples): increase unpinned thread priority in threading-multicore
# Description

Works around scheduler behavior when another higher-priority thread is active (e.g., executor thread).

## Issues/PRs references

Fixes #934.

## Open Questions

## Change checklist

- [ ] I have cleaned up my commit history and squashed fixup commits.
- [ ] I have followed the [Coding Conventions](https://ariel-os.github.io/ariel-os/dev/docs/book/coding-conventions.html).
- [ ] I have performed a self-review of my own code.
- [ ] I have made corresponding changes to the documentation.
closed 2025-03-25 by kaspar030