ariel-os/ariel-os #941: CH32 support

emmanuelsearch, 2025-03-25:
### Description

e.g. support for the [CH32V307VCT6](https://de.aliexpress.com/item/1005007639944954.html)

### Motivation

This is a popular RISC-V alternative.

### Requirements

The network examples work over the ethernet interface of the board.

### Are you planning to do it yourself in a pull request?

Maybe

chrysn, 2025-03-25:
See also the RIOT thread on a similar topic: https://github.com/RIOT-OS/RIOT/issues/20820, where I collected a lot of data already, and Mikolai just linked https://olimex.wordpress.com/2025/03/24/wch-32-bit-risc-v-microcontrollers-and-tools-in-stock/ there.

romainreignier, 2025-05-14:
Just in case you don't know [ch32-rs/ch32-hal](https://github.com/ch32-rs/ch32-hal) has some support for it with Embassy, but no Ethernet support yet for V307 (V208 has a [PR](https://github.com/ch32-rs/ch32-hal/pull/89) for it).