ariel-os/ariel-os #942: Guide to using Visual Studio Code for Ariel OS

emmanuelsearch, 2025-03-25:
### Description

Provide a short how-to support for coding with this editor.

### Motivation

Visual Studio Code is an extremely popular editor...

### Requirements

Provide a short how-to support for coding with this editor.

### Are you planning to do it yourself in a pull request?

No

nponsard, 2025-07-01:
The next step is to have support for developing inside the ariel-os repository.

nponsard, 2025-07-24:
I'm stumbling on an issue where some parts of rust-analyzer still use the resolver version 1, meaning all the target-specific features get enabled, for example in `ariel-os-nrf`:

```toml
[target.'cfg(context = "nrf52840")'.dependencies]
embassy-nrf = { workspace = true, features = ["nfc-pins-as-gpio", "nrf52840"] }
nrf-sdc = { workspace = true, features = ["nrf52840"], optional = true }

[target.'cfg(context = "nrf5340-net")'.dependencies]
embassy-nrf = { workspace = true, features = ["nrf5340-net"] }
nrf-sdc = { workspace = true, features = ["nrf5340-net"], optional = true }
```

With only `--cfg context="nrf5340-net"` in the `RUSTFLAGS`, rust analyzer will still consider that the features `nrf5340-net` and `nrf52840` are enabled on `nrf-sdc` and `embassy-nrf`, leading to errors like "expected 10 arguments, found 6" in `ariel-os/src/ariel-os-nrf/src/ble.rs`.

This issue seems to be tracked in https://github.com/rust-lang/rust-analyzer/issues/15233 . I also created a forum post about it : https://users.rust-lang.org/t/rust-analyzer-seems-to-use-resolver-1/132235/