ariel-os/ariel-os #997: Adding support for the nrf52840dongle

Ollrogge, 2025-04-14:
I would like to add support for the `nrf52840dongle` to ariel-os. Since `probe-rs` doesn’t support programming the `nrf52840dongle`, and tools like `nrfutil` are required instead, what would be the best way to integrate this ?

Ollrogge, 2025-04-28:
👍 I will look into this. Thanks for the pointers !

chrysn, 2025-05-02:
An external debug probe is relatively cumbersome with that board: It merely has a footprint for a 10-pin SMD connector on the bottom, a 10-pin TAG Connect zero-footprint pad (bummer: I have the 6-pin probe), and castellated areas at the neck of the USB connector for SWCLK/SWDIO; nothing immediately useful to most users.

nrfutil is something we do have support in RIOT OS, so we can tap into that experience; sadly, [the Python tool](https://github.com/NordicSemiconductor/pc-nrfutil) is EOL by Nordic (and, worse, doesn't work with recent Python versions, last I checked was 3.10), with the successor being way less accessible (who downloads-and-runs software from a vendor). There is a [rust tool available](https://github.com/ferrous-systems/nrfdfu-rs), even through crates.io (as `nrfdfu`) -- it hasn't been updated in 3y, but then again, that might all suffice, and stems from trustworthy authors. The tool doesn't immediately work with the RIOT built binaries (which I know to work with the old vendor tool), but that may be just due to a false-positive check ("error: firmware starts at address 0x0, expected an address equal or higher than 0x1000 to avoid a collision with the bootloader" when nm says that ROM symbols start at 0x1000).

So far, AFAIK, this would be the first board we'd support a bootloader on (an earlier approach from me on the Particle Xenon boards was too early; https://github.com/ariel-os/ariel-os/pull/23).

chrysn, 2025-05-05:
I've played around a bit more with nrdfu using my [exploration branch](https://github.com/chrysn-pull-requests/nrfdfu-rs/tree/chrysn-exploration) (which will need some cleaning up into PRs). Doesn't immediately help, but has a `--get-versions` switch that gives some output that helps understand the layout of this board:

```
$ cargo run -- --get-versions
* image 0: Bootloader, version 0. Starting 0xe0000, length 122880.
* image 1: Application, version 1234. Starting 0x1000, length 19548.
```

No sure where that's documented, but we do learn from this that there is
* a first stage bootloader in the first page (only place it can be, 0x1000 is right after that)
* the application at the second page all the way up to 0xe0000 (last 32 pages)
* the USB part of the bootloader at 0xe0000, which also stores internally (looked at it: on the last flash page) some information about the available images, eg. the length and version as provided at upload time.

I also learned that the softdevice is treated as a separate image. All devices I have, I already flashed something RIOT based onto, so the softdevices are gone; I'd be curious to see the output on your devices if there's the softdevice still on it.

Ollrogge, 2025-05-06:
> 
> I also learned that the softdevice is treated as a separate image. All devices I have, I already flashed something RIOT based onto, so the softdevices are gone; I'd be curious to see the output on your devices if there's the softdevice still on it.

I also used my dongle to flash RIOT based application on it :/ Therefore, my output looks similar:

```bash
* image 0: Bootloader, version 0. Starting 0xe0000, length 122880.
* image 1: Application, version 1. Starting 0x1000, length 49708.
```

chrysn, 2025-05-20:
Solving https://github.com/ariel-os/ariel-os/issues/639#issuecomment-2894671309 should make this easier; that won't solve the "we have to model the bootloader" aspect (that's #1021), but it would enable having an explicit build context (say, a board named nrf52840dongle-with-bootloader) where the flash settings could be applied.