ariel-os/ariel-os #1023: Question about executor-interrupt & thread

liu0fanyi, 2025-05-11:
### Description

1. what's the priority between this two? since they can independently set their own priorities.
2. How should they communicate with each other? 
    Since I've only seen examples of communication between executor-thread and threads in example:thread-async-interop, 
    does executor-interrupt just use SIGNAL like in thread-async-interop?

### Motivation

no

### Current Implementation

no

### Are you planning to do it yourself in a pull request?

No

kaspar030, 2025-05-12:
Hi @liu0fanyi!

> 1. what's the priority between this two? since they can independently set their own priorities.

The interrupts and thus the interrupt executor(s) have a higher priority than all threads. When an interrupt occurs, it preempts whatever thread is currently running, independent of whether there's an executor running in either of them.

> 2. How should they communicate with each other?

Everything that works for `executor-thread` should also work for `executor-interrupt`. Actually, `examples/thread-async-interop` defaults to `executor-interrupt` for many supported boards.

Feel free to ask more questions! (or close this issue at will)

liu0fanyi, 2025-05-12:
> Hi [@liu0fanyi](https://github.com/liu0fanyi)!  你好！
> 
> > 1. what's the priority between this two? since they can independently set their own priorities.这两个之间的优先级是什么？因为它们可以独立设置自己的优先级。
> 
> The interrupts and thus the interrupt executor(s) have a higher priority than all threads. When an interrupt occurs, it preempts whatever thread is currently running, independent of whether there's an executor running in either of them.中断以及中断执行器（executor）的优先级高于所有线程。当中断发生时，它会抢占当前正在运行的线程，无论该线程中是否有执行器正在运行。
> 
> > 2. How should they communicate with each other?他们应该如何相互通信？
> 
> Everything that works for `executor-thread` should also work for `executor-interrupt`. Actually, `examples/thread-async-interop` defaults to `executor-interrupt` for many supported boards.适用于 `executor-thread` 的所有内容也应适用于 `executor-interrupt` 。实际上，对于许多支持的板子， `examples/thread-async-interop` 默认设置为 `executor-interrupt` 。
> 
> Feel free to ask more questions! (or close this issue at will)随时欢迎提出更多问题！（或随意关闭此 issue）

thanks for your reply！
and another question:
* Should I assume that all current I/O interaction operations should be handled by Embassy tasks, 
* while threads are mainly used to communicate with Embassy tasks and to defer heavier computation tasks?
* Are there any specific scenarios where this kind of coordination would be used? Sorry, I'm still a beginner in this area...

kaspar030, 2025-05-12:
> * Should I assume that all current I/O interaction operations should be handled by Embassy tasks,

Yes, or by threads by using `block_on` on the io futures.

> * while threads are mainly used to communicate with Embassy tasks and to defer run heavier computation tasks?

This makes sense.

> * Are there any specific scenarios where this kind of coordination would be used? Sorry, I'm still a beginner in this area...

Threads have, compared to embassy tasks, a defined priority order. That also might be useful for an application.
Threads are also very useful for imported code that was not built with async in mind.

The rule of thumb would for new code would be: use async by default, use threads when you need them (e.g., for long-running stuff, or for imported non-async code, or to better control priorities).

liu0fanyi, 2025-05-15:
Sorry, one more question. 

If I use the embassy's thread-executor, 
I noticed that ariel-os cannot use multi-threading(in document).
 
My understanding is that the thread-executor itself runs a run->loop where it enters sleep via wfe after each poll done, it can't see if any ariel's thread still runing

and there no method let ariel's thread to wake up it?

So, does this mean that if I use multi-threading—whether with a custom executor or the interrupt-executor
I no longer have a sleep and wake-up (wfe/sev or wfi)mechanism?

kaspar030, 2025-05-19:
I'll close this, feel free to re-open if there are more questions.

closed 2025-05-19 by kaspar030