ariel-os/ariel-os #1038: rtt buffers too large

kaspar030, 2025-05-19:
enabling rtt&defmt configures a 1KiB buffer each. On more RAM constrained devices, that's just too much.

1. drop the plain output channel completely if defmt is used? (meaning, drop `print` support in favor of log and just `println()`, which defmt supports?

2. reduce buffer size, or make configurable?

chrysn, 2025-05-20:
Dropping the plain-text channel would make us a worse stress test for probe-rs ;-) -- but realistically, I think yes we should drop that print-only channel, and have a println and dbg that go through defmt if that works.

As for buffer size, do we have any easy way of observing how close we're coming to an overflow, and/or whether an overflow happened? (From the host side, that is.)

chrysn, 2025-05-20:
It's not only about how close we get to overflowing at runtime, it's also about whether or not anything was lost at startup. The RPC approach that they're adding into probe-rs could allow us some more control eventually.

kaspar030, 2025-05-20:
Doesn't 'probe-rs run' keep the MCU stopped until probe-rs re-attaches? (Could it?)

chrysn, 2025-05-20:
Good point, I don't know if it does; it can definitely not be easy, given that it has to stop it past CRT0 to even see the RTT region in RAM (if that's how it works).