ariel-os/ariel-os #1115: Support the modem on nRF9151

nponsard, 2025-06-27:
### Description

Add support to the modem on the nRF9151 sip. This modem is only officially usable via the nordic SDK and includes a closed source binary blob. 

There is one project aiming to support the modem for Rust: [nrf-modem](https://github.com/diondokter/nrf-modem).
There is also another project centered around the nrf9160 that doesn't seem to get much updates as the company's website is offline now: [nrfxlib](https://github.com/nrf-rs/nrfxlib).

According to the documentation, using the modem will require the application core to run in NS mode, meaning isolation features will be enabled. Ariel OS doesn't seem to be able to boot in NS mode currently.

### Motivation

We have some upcoming projects using the thingy91x (nRF9151) that will need access to the GPS and LTE-M features.

### Requirements

1. Access to GPS features to obtain a position
2. Access to LTE-M features to do networking

### Are you planning to do it yourself in a pull request?

Yes

nponsard, 2025-07-01:
I managed to go a bit forward in the initialization of the modem in NS mode, I get some logs back (about the TX region, RX region, Trace region) but then the initialization fails with the error ` NrfError(-116)`. This seems to be similar to https://github.com/diondokter/nrf-modem/issues/21.

Also it seems that the executor-thread doesn't work in NS mode currently, the firmware crashes in `ariel-os-rt` at `threading::start();` with the red led of the thingy91x lighting up.

diondokter, 2025-07-02:
> There is one project aiming to support the modem for Rust: [nrf-modem](https://github.com/diondokter/nrf-modem).
> There is also another project centered around the nrf9160 that doesn't seem to get much updates as the company's website is offline now: [nrfxlib](https://github.com/nrf-rs/nrfxlib).

To give a little bit of background: First there was nrfxlib and Jonathan made things work. But then it sort of got abandoned. I think I updated it once or so, but then decided this shiny new async thing was very interesting. So I made my own version which is nrf-modem. nrf-modem is very much inspired from nrfxlib.

nrfxlib is no longer supported (and is quite old w.r.t. the blob version it uses). Nrf-modem is sort of supported by me, passively. That means I'm not really doing active work on it, but will respond to issue and merge new work.

Nrf-modem is made to work with bogstandard embassy. But I'd be open to changes to make it work in more places!

nponsard, 2025-07-02:
Thanks for the clarification! 
I'm trying to track down where the problems I'm having comes from, once I figure this out I'll open an issue/PR if changes need to be made.

nponsard, 2025-07-04:
I also went back at trying to make the modem work in S mode, without TF-M. I'm currently trying to make it work first in a simple embassy project, but I cannot manage to have it run.

Here is my attempt: https://github.com/nponsard/nrf-modem-test-s-mode

This seem to be linked to [this comment](https://github.com/diondokter/nrf-modem/issues/32#issuecomment-2704598018) but I didn't understand how to "restore those values".