ariel-os/ariel-os #1143: monitor external dependencies to detect known vulnerabilities

jkarinkl, 2025-07-14:
[OpenSSF best practices silver level](https://www.bestpractices.dev/en/projects/10610?criteria_level=1), under Quality, 'Externally-maintained components':
- Projects MUST monitor or periodically check their external dependencies (including convenience copies) to detect known vulnerabilities, and fix exploitable vulnerabilities or verify them as unexploitable.
Extra info: This can be done using an origin analyzer / dependency checking tool / software composition analysis tool such as [OWASP's Dependency-Check](https://www.owasp.org/index.php/OWASP_Dependency_Check), [Sonatype's Nexus Auditor](https://www.sonatype.com/nexus-auditor), [Synopsys' Black Duck Software Composition Analysis](https://www.synopsys.com/software-integrity/security-testing/software-composition-analysis.html), and [Bundler-audit (for Ruby)](https://github.com/rubysec/bundler-audit). Some package managers include mechanisms to do this. It is acceptable if the components' vulnerability cannot be exploited, but this analysis is difficult and it is sometimes easier to simply update or fix the part.

Todo:
- [ ] Install Dependabot to fix this

ROMemories, 2025-07-15:
A benefit of Renovate is that it's not limited to GitHub, unlike Dependabot (at least the last time I checked), and that would also us to migrate more easily if need be.

anlavandier, 2025-07-15:
> A benefit of Renovate is that it's not limited to GitHub, unlike Dependabot (at least the last time I checked), and that would also us to migrate more easily if need be.

Gitlab seems to have the same kind of thing with their own [Gitlab-Dependabot](https://gitlab.com/dependabot-gitlab/dependabot).