ariel-os/ariel-os #1242: Bumping MSRV to 1.87+

ROMemories, 2025-08-27:
This issue is intended to keep track of what we could do after bumping the MSRV to 1.87+:

- [ ] Updating `sequential-storage` to v4.0.2+, see #1240

---

MSRV bump checklist:

The MSRV in the README and the [`rust-version` manifest key](https://doc.rust-lang.org/cargo/reference/rust-version.html) must be updated atomically (in the same PR).

anlavandier, 2025-08-27:
I think that would also need to change: 
- The nightly version pinned in [`laze-project.yml`](https://github.com/ariel-os/ariel-os/blob/main/laze-project.yml#L864) because it's the one [grepped by our script](https://github.com/ariel-os/ariel-os/blob/main/scripts/rust-toolchain.sh) and [used in CI](https://github.com/ariel-os/ariel-os/blob/main/.github/workflows/build.yml#L184)
- The version of the `dtolnay/rust-toolchain` action to 1.87+ [here](https://github.com/ariel-os/ariel-os/blob/main/.github/workflows/build.yml#L193)
- Maybe change the version used for esp-rs [here](https://github.com/ariel-os/ariel-os/blob/dc756ae143f8542de508dc824c336be8c187c8a9/.github/workflows/build.yml#L205)

anlavandier, 2025-12-03:
Does the assignement mean that you want me to make all of the changes listed in this issue ?